5axes/Calibration-Shapes #1: Use postprocessing scripts without copying them

fieldOfView, 2021-01-18:
It should be possible to use your postprocessing scripts without first having to copy them over and restart Cura. Cura should be able to find the scripts from inside your plugin folder if you move your `scripts` folder to into a folder named `resources`, and add this to your plugin initialisation:

`Resources.addSearchPath(os.path.join(os.path.dirname(os.path.abspath(__file__)), "resources"))`

This will add that resources folder to the paths Cura checks for script files.

5axes, 2021-01-18:
Unfortunatly no it doesn't work. Your solution is correct but as the PostProcessingPlugin is launched before the cutom plugin, even if the code is correct the scripts are not loaded byt the PostProcessingPlugin  via the function  : def loadScripts(self, path: str) -> None:

2021-01-18 14:36:**53,431** - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PostProcessingPlugin
...
2021-01-18 14:36:**54,382** - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleShapes

fieldOfView, 2021-01-18:
It doesn't work, but that is not because the PostProcessingPlugin loads before your plugin, but because the PostProcessingPlugin does not use the resource search paths correctly :-(

https://github.com/Ultimaker/Cura/blob/4.8/plugins/PostProcessingPlugin/PostProcessingPlugin.py#L145
Instead of using `getStoragePath`, it should be using `getAllPathsForType`. Oh well, it was worth the try.

nallath, 2021-01-19:
Whoops, that's our bad :(

fieldOfView, 2021-01-19:
It just might be my bad ;-)

https://github.com/nallath/PostProcessingPlugin/commit/12d4d177b624272a0083b0ecc17f737c3c19ca1d#diff-29915b71ed69033c978493b7def0019f64e3064ac80c4ab127d74ac5d3c3628bR159

fieldOfView, 2021-01-19:
Here's a fix: https://github.com/Ultimaker/Cura/pull/9147

nallath, 2021-01-19:
> It just might be my bad ;-)
> 
> [nallath/PostProcessingPlugin@12d4d17#diff-29915b71ed69033c978493b7def0019f64e3064ac80c4ab127d74ac5d3c3628bR159](https://github.com/nallath/PostProcessingPlugin/commit/12d4d177b624272a0083b0ecc17f737c3c19ca1d#diff-29915b71ed69033c978493b7def0019f64e3064ac80c4ab127d74ac5d3c3628bR159)

Well, we were still paying you at that time, so it's our fault :P

fieldOfView, 2021-01-19:
Nope, this was a year and 2 days after my last day on the team... But it was your fault for not spotting it in the review ofcourse. It is always your fault.

nallath, 2021-01-19:
>  It is always your fault.

Damn right!  :D

nallath, 2021-02-12:
I just merged the PR.

5axes, 2021-06-02:
Ok I have modified the source code V1.3.0. Don't need to use the Copy scripts function anymore. Have been removed from the men.

closed 2021-06-02 by 5axes