5axes/Calibration-Shapes #4: Adding the calibration objects doesn't wor on Linux

nallath, 2021-01-27:
I'm testing the plugin on Fedora and the second set of calibration objects (Calibration cube, temp tower, etc) are not loaded when clicked.

The logs show the following:
```
2021-01-27 09:31:07,436 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]: Traceback (most recent call last):
2021-01-27 09:31:07,438 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/home/jaime/Development/Uranium/UM/Qt/Bindings/ExtensionModel.py", line 54, in subMenuTriggered
2021-01-27 09:31:07,440 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     item["extension"].activateMenuItem(option_name)
2021-01-27 09:31:07,442 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/home/jaime/Development/Uranium/UM/Extension.py", line 50, in activateMenuItem
2021-01-27 09:31:07,444 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     self._menu_function_dict[name]()
2021-01-27 09:31:07,445 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/home/jaime/.local/share/cura/4.7/plugins/CalibrationShapes/CalibrationShapes/CalibrationShapes.py", line 212, in addTempTower
2021-01-27 09:31:07,446 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     self._addShape(self._toMeshData(trimesh.load(model_definition_path)))
2021-01-27 09:31:07,448 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/usr/local/lib/python3.8/site-packages/trimesh/exchange/load.py", line 108, in load
2021-01-27 09:31:07,449 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     ) = parse_file_args(file_obj=file_obj,
2021-01-27 09:31:07,452 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/usr/local/lib/python3.8/site-packages/trimesh/exchange/load.py", line 573, in parse_file_args
2021-01-27 09:31:07,454 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     raise ValueError('string is not a file: {}'.format(file_obj))
2021-01-27 09:31:07,455 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]: ValueError: string is not a file: /home/jaime/.local/share/cura/4.7/plugins/CalibrationShapes/CalibrationShapes/models\TempTower.stl
```

baikal, 2021-01-27:
@nallath Just noticed the same problem. This is due to the Windows-style path separator \ in the .stl names.
Using the platform-independent path joining instead fixes that. E.g.
`model_definition_path = os.path.join(os.path.dirname(os.path.abspath(__file__)), "models", "CalibrationCube.stl")`
instead of
`model_definition_path = os.path.join(os.path.dirname(os.path.abspath(__file__)), "models\CalibrationCube.stl")`
in CalibrationShapes.py. And the other six ones as well ...

5axes, 2021-01-27:
Solved V1.0.4 thanks a lot @baikal

closed 2021-01-27 by 5axes