5axes/Calibration-Shapes #11: [Feature Request] PETG Temp Tower

hawkeyexp, 2021-02-07:
Hi,

first to say: great plugin!!!

Would be nice if you can add more temp towers for different materials - but i think the big three pla / abs / petg is the most needed - so only petg is missing :-)

5axes, 2021-02-08:
Updated in V1.0.8

userosos, 2021-03-07:
I add the tower - part for calibration/PETg temp tower. Add chage g-code Tempfan tower. And it not work.

[Untitled.zip](https://github.com/5axes/Calibration-Shapes/files/6097311/Untitled.zip)

5axes, 2021-03-07:
Looks like you set the wrong parameter in the script .. When we log your GCode change are made every 3 steps.
[image]

As your settings for the layer Height are set to : 
- First layer height : 0.3
- Layer height : 0.2

[image]

In this case settings must be : 

[image]

And you should get these result :  
[image]

userosos, 2021-03-07:
I use default settings
[image]

5axes, 2021-03-07:
Ok that's normal you should not use the Fan option, that's why you have the same temperature on 3 level. Read the wiki concerning the script options

userosos, 2021-03-08:
> 
> 
> Ok that's normal you should not use the Fan option, that's why you have the same temperature on 3 level. Read the wiki concerning the script options

It works! Thx!

5axes, 2021-05-16:
Updated in V1.0.8

closed 2021-05-16 by 5axes