5axes/Calibration-Shapes #15: [Feature Request] Nozzle Offset Calibration (dual/multi nozzle)

rubienr, 2021-02-24:
Hi, 

Thanks for the very convenient plugin. I appreciate your work.

Is it possible to provide a calibration pattern for dual (or multi) nozzle print head? It would be very convenient to have a pattern that helps me finding out the correct X and Y offset of the second nozzle.

**The Problem**

I built a very accurate switching extruder printhead but the mounted stock hotend seems to have a lower dimensional accuracy resulting in a serious offset (for example  x_offset=+1mm, y_offset=-0.5mm). The real offset I cannot measure in the CAD drawing and instead of measuring with the caliper (and probably damaging/scratching the brass tip hole) a calibration pattern would be very convenient. Such a pattern could be inspired by the color ink printer calibration test.

**Implementation Suggestion**

The calibration pattern should provide a few pairs of X and Y aligned lines (or pads for better adhesion). The two neighboring aligned pairs are printed with T0 and T1 each. There should be more such pads per axis with a slightly incremented/decremented offset in their alignment, so that I can choose the correct one.

**Usage Example**

1. I set the assumed (as of CAD drawing) X and Y nozzle offset in the printer firmware.
1. I load the calibration pattern in Cura and use a post processing script for manipulating the X/Y offset parameters.
1. I start the calibration pattern printout.
1. I search for the best matching pairs in X and then in Y direction.
1. I update the offset in firmware with the new offset parameters.
1. I start a second printout for verification.

5axes, 2021-02-24:
Could be a nice feature, But As I'm not a dual extruder user it's not so easy for me to integrate the good part.  But could be something like this [solution ](https://docs.aon3d.com/calibration/xy_tool_calibration/)?

5axes, 2021-02-24:
Solution B ... seems to be the easyest to use and program ( no need a special file for every type of nozzle )

rubienr, 2021-02-24:
Solution B could do the trick. I can help with printing/debugging (find me on DC #8902).

5axes, 2021-02-24:
You can test the CalibrationMultiExtruder  by downloading the STL files  Red and White
[https://github.com/5axes/Calibration-Shapes/blob/master/models/CalibrationMultiExtruderWhite.stl](https://github.com/5axes/Calibration-Shapes/blob/master/models/CalibrationMultiExtruderWhite.stl)
[https://github.com/5axes/Calibration-Shapes/blob/master/models/CalibrationMultiExtruderRed.stl](https://github.com/5axes/Calibration-Shapes/blob/master/models/CalibrationMultiExtruderRed.stl)

And bicolorcube
[https://github.com/5axes/Calibration-Shapes/blob/master/models/CubeBiColorRed.stl](https://github.com/5axes/Calibration-Shapes/blob/master/models/CubeBiColorRed.stl)
[https://github.com/5axes/Calibration-Shapes/blob/master/models/CubeBiColorWhite.stl](https://github.com/5axes/Calibration-Shapes/blob/master/models/CubeBiColorWhite.stll)

[image]

rubienr, 2021-02-27:
Hi,

I did not test your pattern yet, but tried to do another prove of concept [1]. The print time seems to be significantly shorter than the tower. I created two neighbouring paratmetrize-able combs where the center teeth are aligned. The one's teeth has an offset of `2mm` the others' teeth offset is `2.2mm` (same principle as the caliper). With that comb I can immediately spot the offset within a rage of `+/-2mm` by finding the teeth that perfectly match. Ideally its the two in the center, but in my case it was not.

**The two approaches:**
1. comb on top of the initial layer (takes about 15')
2. comb directly on build plate (takes far less)

[image]
(Sorry for the black taint on the right example, the gray filament ran out during print.)
[image]

**How to parametrize (FreeCad):**
1. open the FCStd file
2. go to the "vars" spread sheet
3. modify parameters according to your needs
4. click one of the two fusion objects and press `Ctrl + e` to export to `.stl`
5. repeat for the 2nd fusion object

[image]
 
**Notes**

I did not play around with the parameters that much but I think the print time can be still slightly enhanced. In my setup the calibration pattern was done in one layer (max two layers if initial layer is needed for adhesion).
However, the initial layer takes the most but it has to be considered that not every printer is capable of printing single lines directly on the build plate without it. Probably by removing the prime tower brim and with a wall count of one the pattern can be printed faster.

**My printing conditions were**
 - Nozzle Size : `0.4mm`
 - Layer Height : `0.3mm`
 - Initial Layer Height : `0mm` without initial layer, `0.3mm` otherwise
 - Line Width : `0.4mm`
 - Wall Line Count : `2`
 - Top/Bottom Thickness : `0mm` without initial layer, `0.8mm` otherwise
 - Initial layer speed: `40 mm/s`
 - Wall speed `80mm/s`
 - Enable Retraction : `True`
 
[1] https://github.com/rubienr/cad/blob/master/creality-ender-5-plus/printer-test-objects/nozzle-to-nozzle-offset-calibration-pattern/00.stl

rubienr, 2021-02-27:
An OpenSCAD script for the afore mentioned pattern can be found at [1]: 
```
openscad nozzle-to-nozzle-xy-offset-calibration-pattern.scad
```
or try various parameters:
```
openscad -D var1=val1 -D var2=val2 ... nozzle-to-nozzle-xy-offset-calibration-pattern.scad
```

**Arguments**

* `teeth_count` number of teeth
* `tooth_width` width of one single tooth
* `tooth_height` height of teeth / comb
* `tooth_length` length of one single tooth
* `teeth_gap` space in between teeth
* `comb_clearance` comb to comb distance
* `base_height` comb base (optional), set to 0 to disable
* `render_mode` 0 ... all, 1 for nozzle 1, 2 for nozzle 2
[image]

[1] https://github.com/rubienr/cad/blob/master/creality-ender-5-plus/printer-test-objects/nozzle-to-nozzle-offset-calibration-pattern/nozzle-to-nozzle-xy-offset-calibration-pattern.scad

5axes, 2021-02-28:
I will have a look to your file. Is it possible to find some set of parameter to create one design adapted to various size of  nozel 0.4, 0.6 0.8 ?

rubienr, 2021-03-01:
I think this is doable. If we say the slice shall be done with a wall count of two, then for a 0.8mm nozzle the comb teeth need to be wider (probably: `2*2*0.8 + some clearance`). The teeth gaps can remain the same. I believe the more complicated part is the layer height. I intended to make this pattern print-able in two layers, but I don't know how thin a layer can be printed with a 0.8mm nozzle. I will try a test print with a larger layer height of 0.4mm with a 0.4mm nozzle. Probably a 0.8mm nozzle can go that thin, then we can meet in the middle: 0.4mm layer height (each: tooth height and base height). 

For the moment I would try:
`openscad -D tooth_width=3.5 -D tooth_height=0.4 -D base_height=0.4 nozzle-to-nozzle-xy-offset-calibration-pattern.scad`

Slicer settings:

* Nozzle Size : `0.4mm` to `0.8mm`
* Layer Height : `0.4mm`
* Line Width : `0.4mm` to `0.8mm` (shall be the nozzle diameter)
* Wall Line Count : `2`
* Top/Bottom Thickness : `0mm` without initial layer, `0.4mm` otherwise
* Initial layer speed: `40mm/s` (or lower without initial layer; depending on your printer's adhesion capabilities)
* Wall speed: up to `80mm/s` (lower speed may produce clearer corners)
* Enable Retraction : `True`

5axes, 2021-03-01:
Would be also Great to get the X 0 Y 0 in the "middle" of the total geometrie and Z=0 for the lower surface in order to put it directly in the center of the build plate.

closed 2021-05-16 by 5axes