5axes/Calibration-Shapes #16: Help with Retraction tower

eblieb, 2021-02-27:
Little confused on the retraction tower script. How exactly are we suppose to config the values of the post processing script?

5axes, 2021-02-28:
Did you have a look to the associated Wiki page : [Retract-Towe](https://github.com/5axes/Calibration-Shapes/wiki/Retract-Tower) ?

eblieb, 2021-03-02:
Thanks I missed that one.

SputnikOne, 2021-03-12:
I also have a question about the Retract-Tower documentation. For the two Post-processing script settings. Are you meant to create two separate Retract Tower scripts? Otherwise, if I select speed change the settings, then select the drop-down for distance - the speed settings are not maintained if I were to select speed again.

5axes, 2021-03-12:
You should create two projects. One for the speed, one for the distance. Or set the value for the speed, test the result on your printer, then go back in the project , change the settings in distance and generate a new iso file.

wessold, 2022-02-07:
I am trying to set me retract. on the wiki it talks about stepping the retract value but when I go to add the script it has speed and distance. This might be a stupid question. But, I am now to 3d printing. 

[image]

[image]

Billy-d-Kidd, 2022-03-23:
Hello! I'm new to this and was wondering do you only need to add the script once, or is it like a temp tower where you add the script several times with different values then print?

5axes, 2022-03-23:
You need to use just one Script.   These scripts are done to replace the solution to use several ChangeatZ Script

Billy-d-Kidd, 2022-03-24:
Went back and tried the script, but I'm obviously either not doing something right, or something else is going wrong.
[image]
The tower in front was my first test, and it shows to be fine all the way up, no strings or oozing, but as you can see on the test print next to it, it is full of oozing.
So I went back and printed the test script a second time, it comes up fantastic with no oozing or stringing, but the test print next to it, has oozing and some stringing in it as well.
Is there a step I'm missing, or what?
I also tried the flow rate script, and when it was done all of the pegs fit into the slots, but the bottom one is now stuck, but the rest slide out with no problem.

5axes, 2022-03-24:
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/stringing.md

Billy-d-Kidd, 2022-03-24:
Thanks, that's some helpful info there.

lukamicoder, 2023-05-14:
I have the same issue as Billy-d-Kidd had: I fine tuned the retraction speed and distance by printing the tower a few times and eventually I was able to print a near flawless tower (with 7mm retraction distance and 50mm/s speed). However, when I printed a couple of different retraction test prints (same filament, same settings) they were all stringy. I read that page and it is useful. Nevertheless, it doesn't explain why the settings that work the best for the tower don't seem to work for other prints.  After all isn't the whole point of this tower to get the retraction settings that can be used for other prints with the same filament? What am I missing here?

5axes, 2023-05-14:
Retraction tower is a test part to find some settings. But it's not the innsurence to get the best settings. You must test your settings on an other test part like the : https://github.com/5axes/Calibration-Shapes/wiki/Retract-Test-part

closed 2023-05-14 by 5axes