5axes/Calibration-Shapes #17: Store models in binary stl instead of ascii stl

nallath, 2021-03-01:
As per title. Binary STL is much smaller than ascii stl. It won't make that much differnce for the small models, but it does add up for the larger ones.

5axes, 2021-03-01:
Yes you are right but as I want to keep OpenScad as reference  (like that anyone would be able to modify the design and why not suggest a better design :  shorter or more revelant for the test)  and As OpenScad Only generate an Ascii STL file ( or I'm missing something in the options of OpenScad).  

It could be a good solution if in cura It makes a big difference to load an ascii or a binary file via the trimesh library, but the test I've made doesn't seems to show any significant difference.

5axes, 2021-03-01:
Finaly it's a not a bad idea ... Drag and Drop in Cura and Export of the STL file will offer the possibility to reset automaticaly the position to 0 , 0 .. I Will convert the file for the next release

5axes, 2021-03-01:
STL file Converted to STL binary except for Dual extruder sample part.

closed 2021-03-01 by 5axes