5axes/Calibration-Shapes #19: Temperature tower does not work with nozzle size of 0.8. it does not slice properly.

AlexZan, 2021-03-20:


5axes, 2021-03-20:
It s designed for a 0.4 nozzle, if you want to use it with a 0 8 nozel change the size by using the Cura scalling function

MarphXL, 2021-05-16:
Perhaps you could give us a config-attribute for nozzle-size? It would be great. And thanks for your work: amazing: I love this plugin

5axes, 2021-05-16:
Can you developped what do you mean by "config-attribute" ?

MarphXL, 2021-05-16:
Hi, in the cura-menu you have a menu-option "define default size". 
Perhaps you can add a menu-option "nozzle size"

5axes, 2021-05-16:
In this case we can directly get this information from the Cura settings 
[image]

But this is not the real problem ...  If I add a design for let say the most common nozzle Size ...  0.2  0.4  0.6  0.8 and Why not 1.2 that's a uge number of part for just a limited number of user .   I think it's a simpler solution to keep the same geometry and if someone wants to use a 0.8 nozzle it can do it just by activating the option "Remove all holes"  in the Mesh fixe section. 
[image]

And in this case the actual geometry can be also be used with a 1.2 nozzle diameter.
[image]

5axes, 2021-06-08:
Solved in V1.3 by changing automaticaly  the <code>Remove All Holes</code> in the settings in such case

closed 2021-06-08 by 5axes