5axes/Calibration-Shapes #21: Modified G-Code not included...

Benvolio28, 2021-03-21:
Hi!  Love the tools, however for the temperature tower, the modified G-Code is not part of the project.  The temp just stays the same as my cura settings. I'm on an Ender 3 V2 (running the Smith 3D software), with Cura 4.8.0.

I have installed the plug in
I have added the modified G-Code to Cura
I have restarted Cura.

Thanks for your help, and thank you for the plugin!

5axes, 2021-03-22:
There is no modified G-code in this plugin. I don't really understand what kind of issue you are talking about ?

Benvolio28, 2021-03-22:
With the temp tower, it does not alter the temp for each step.  It stays at
the temp set in my cura settings. So, I usually print pla at 200... So,
it's always 200.  No variance.

Thanks for your reply.

5axes, 2021-03-22:
did you add the TempFan postprocessing script and configure it ?

AlexZan, 2021-03-22:
I have the same issue, I have added the TempFan script and there is only one single m109 command in the gcode

Benvolio28, 2021-03-29:
So sorry it took me a bit to get back to you... work got in the way.
Yes, I did add the TempFan script and configure it

5axes, 2021-03-29:
Ok i haven't correctly read your first message. I don't know what is this "Smith 3D software"  the script are designed for a Marlin compatible printer. Could be the origine of your issue ?

XAMCTBO, 2021-05-03:
guy`s I have totally same issue like guy who create subject. Cura 4.8 adding this plugin, restart cura, adding modifyed G code  setting up layers and start temp, push slice -> send to octoprint - temperature not changing.... 
Octopi updated 
cura connected to octoprint(latest)

5axes, 2021-05-25:
Whitout news .. don't need to keep it open

closed 2021-05-25 by 5axes