5axes/Calibration-Shapes #24: import error after installation on Ubuntu 20.04

Acer54, 2021-03-25:
Hi, 
dont know if it is plugin related issue or maybe from cura dependencies.
After installing your (very useful) plugin in Cura 4.8 on Ubuntu, I cannot find it in the menu bar, after searching the logs I found this:

> 2021-03-24 22:12:31,502 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [507]: Loaded plugin LocalContainerProvider
2021-03-24 22:12:31,504 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]: Exception: Import error loading module CalibrationShapes
2021-03-24 22:12:31,504 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]: Traceback (most recent call last):
2021-03-24 22:12:31,504 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 667, in _findPlugin
2021-03-24 22:12:31,505 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:     module = imp.load_module(plugin_id, file, path, desc) #type: ignore #MyPy gets the wrong output type from imp.find_module for some reason.
2021-03-24 22:12:31,505 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "/usr/lib/python3.8/imp.py", line 244, in load_module
2021-03-24 22:12:31,505 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:     return load_package(name, filename)
2021-03-24 22:12:31,505 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "/usr/lib/python3.8/imp.py", line 216, in load_package
2021-03-24 22:12:31,505 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:     return _load(spec)
2021-03-24 22:12:31,506 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "<frozen importlib._bootstrap>", line 702, in _load
2021-03-24 22:12:31,506 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
2021-03-24 22:12:31,506 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
2021-03-24 22:12:31,506 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
2021-03-24 22:12:31,506 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "/home/matthias/.local/share/cura/4.8/plugins/CalibrationShapes/CalibrationShapes/__init__.py", line 4, in <module>
2021-03-24 22:12:31,507 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:     from . import CalibrationShapes
2021-03-24 22:12:31,507 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:   File "/home/matthias/.local/share/cura/4.8/plugins/CalibrationShapes/CalibrationShapes/CalibrationShapes.py", line 25, in <module>
2021-03-24 22:12:31,507 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]:     import trimesh
2021-03-24 22:12:31,507 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [669]: ModuleNotFoundError: No module named 'trimesh'
2021-03-24 22:12:31,507 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [739]: Could not find plugin CalibrationShapes

I had to install trimesh by hand (sudo pip install trimesh). After installation and reload of Cura, your plugin was listed in the menue bar successfully.
Dont know if you have influence to this behavior...

5axes, 2021-03-26:
Certainly dame issue as reported in this discussion : https://github.com/5axes/Calibration-Shapes/discussions/23
. Unbutu 20.04 Cura issue, as I don't have this Os I cannot help you but I will say Cura upstream issue in this case.

Milszus, 2021-05-19:
Thanks to @Acer54 for this hint, after "sudo pip install trimesh" it workes for me (Ubuntu 20.04)