5axes/Calibration-Shapes #26: Copy Scripts does not work

pierremilcent, 2021-04-08:
After updating Calibration Scripts to 1.2.2 the copy of scripts does not work : seems related to a sub directory as shown on screenshots

[image]

[image]

5axes, 2021-04-08:
1.2.2 ??? last release 1.2.1 ?

pierremilcent, 2021-04-08:
Oups .... latest installe is 1.2.0 despite the market place is still behind this version ...

[image]
[image]

5axes, 2021-04-08:
try with the 1.2.1

5axes, 2021-04-08:
try with the 1.2.1

pierremilcent, 2021-04-08:
It seems to be solved with this version. 2 scripts updated (retracttower and speedtower).

[image]

5axes, 2021-04-08:
Ok the Scripts\Scripts displayed in the final message is just an Error in the messsage text not in the real destination Folder.  Solved in 1.2.2

closed 2021-04-08 by 5axes