5axes/Calibration-Shapes #28: Issue with Retraction Tower on Cura 4.8.0 / gcode "G1 EXX" lines?

lwiart, 2021-04-18:
Hi guys,

Thank you sooooo much for this amazing plugin where we find all calibration shapes instead of going here and there to get each and every one of them :)

I'm a beginner (got a CR10S Pro v2) a week ago, and using Cura 4.8.0. So I'm pretty sure what I'm describing is just because of my lack of experience/comprehension in 3D printing/Cura/gcode...

So: 
- I calibrated by Sunlu PLA+ at 210° with a temp tower
- Then I went on with the retraction tower : test from 3mm to 7mm retraction
[image]
- After 30mn, here is what I get
[image]
- Surprise! I checked the gcode, and I don't understand how retraction is set on all the G1 lines: 
  - First, I was expecting lines such as "G1 E-X" (where X goes from 3 to 7), but all I can see is 
    `Line 45: G1 F2700 E-7
    Line 49147: G1 E-2 Z0.2 F2400 ;Retract and raise Z`
- Second, each G1 line has a slight E increment, and here is an excerpt of layer 0, and last layer: it starts at E0.03135, and ends at E1379.74101

`
;LAYER:0
M107
G0 F6000 X125.272 Y136.534 Z0.2
;TYPE:SKIRT
G1 F2700 E0
G1 F1200 X126.013 Y135.959 E0.03135
G1 X126.8 Y135.449 E0.0627
G1 X127.628 Y135.008 E0.09406
G1 X128.49 Y134.639 E0.1254
G1 X128.991 Y134.463 E0.14315
G1 X129.136 Y134.416 E0.14825
G1 X130.04 Y134.165 E0.17961
G1 X130.962 Y133.992 E0.21096
...;LAYER:196
;TYPE:WALL-OUTER
;MESH:RetractTower
G1 F2700 E1377.47814
G1 F1229.6 X162.757 Y152.587 E1377.49628
G1 X163.288 Y152.474 E1377.51438
G1 X163.786 Y152.252 E1377.53256
G1 X164.227 Y151.932 E1377.55073
G1 X164.59 Y151.528 E1377.56884
`

Again, I'm pretty sure it's my lack of knowledge, so excuse me for bothering you as it's very certainly a lack of understanding on my side...

[RetractionTower3-7.zip](https://github.com/5axes/Calibration-Shapes/files/6331091/RetractionTower3-7.zip)

5axes, 2021-04-18:
I think I still have an issue when the Value of the Change Layer Offset. Anyway with this model the value should be set to 4 . 
[image]

lwiart, 2021-04-18:
Thanks 5axes! I'll do keep "Change Layer Offset = 4"

In the interim, I tried to generate gcode (Cura 4.8.0) with distance/3/1/38/4, and I still get those weird extrusion values all along the G1s, that go from 0 to 1361. Is that normal?

```
;LAYER:0
M117 speed (3.0/1.0)
M107
G0 F6000 X125.272 Y136.534 Z0.2
;TYPE:SKIRT
G1 F2700 E0
G1 F1200 X126.013 Y135.959 E0.0312
G1 X126.8 Y135.449 E0.06239
G1 X127.628 Y135.008 E0.09359
G1 X128.49 Y134.639 E0.12478
G1 X128.991 Y134.463 E0.14244
G1 X129.136 Y134.416 E0.14751
G1 X130.04 Y134.165 E0.17871
G1 X130.962 Y133.992 E0.20991
...  
;LAYER:196
M117 speed F480
;TYPE:WALL-OUTER
;MESH:RetractTower
G1 F600 X162.757 Y152.587 E1359.32512
G1 X163.288 Y152.474 E1359.34317
G1 X163.786 Y152.252 E1359.36131
G1 X164.227 Y151.932 E1359.37943
G1 X164.59 Y151.528 E1359.3975
G1 X164.861 Y151.056 E1359.4156
G1 X165.029 Y150.541 E1359.43362
G1 X165.086 Y149.998 E1359.45178
G1 X165.029 Y149.457 E1359.46987
G1 X164.86 Y148.941 E1359.48793
```

5axes, 2021-04-18:
Yes it's normal .. You have two modes to controle the Extruder axis : 
The default value are set to absolute mode this is the case in your G-Code . 
You can also switch to an incremental value (Relative Extrusion), In this case you should find the retraction value directly in your Gcode.

[image]

5axes, 2021-04-18:
I see your Error , your configuration of the CR10S is not correct. And a G83 code is missing before your the 
G1 E-2 Z0.2 F2400 : 
[image]

5axes, 2021-04-18:
I recommand to use the following code : 

`M83 ; E Relative`
`G1 F2400 Z20 E-5 ; Retract and Raise Z`

But rectract and Raise the Z on the same line is not a good solution because you are using the same speed for E and Z So you should Change to : 

`G91 ; Set coordinates to relative`
`G1 F1800 E-5 ; Retract filament 5 mm to prevent oozing`
`G1 F3000 Z20 ; Move Z Axis up 20 mm to allow filament ooze freely`
`G90 ; Set coordinates to absolute`

Cura in this case will change the -5 value to the right settings even if you don't specify the M83 instruction

lwiart, 2021-04-18:
Thanks for the hints. It's useful for comprehension.
But since I use Cura,  when I used your plugin to import a Retraction Tower model, then the "RetractTower" script, the gcode snippets I pasted here are from what is being generated (with "Relative Extrusion=FALSE"), to get the best retraction setting between 3 and 7mm.

So, should what you mention be put in the start gcode for the machine? or should the RetractTower.py script be updated?

5axes, 2021-04-18:
These comments are not related to the RetractTower.py script. I think the problem exists more generally on your prints. The changes are related to the configuration of the machine. I am surprised that this problem is not more often encountered for CR10S Pro users

lwiart, 2021-04-28:
Sorry for my belated reply. Thanks for the hint!

5axes, 2021-05-25:
Whitout news .. don't need to keep it open

stephentt, 2021-06-02:
I am also having this issue and I am using the default Ender 3 machine profile in Cura.  I am wondering why this isn't happening to everyone who is using your script on a Creality machine?  Here is the end gcode as default provided by Cura:

'''
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
'''

Here is the end gcode after adding your script:
'''
G91 ;Relative positioning
G1 F2700 E1292.14860
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y220 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
'''

So like you mentioned above, it is a G91 change to relative followed immediately by an absolute extrude value as modified by your script.  So it seems like the best thing would be for the script to leave the start and end gcode unmodified if there is a way to identify it separately from the rest of the gcode?

5axes, 2021-06-02:
Modification of the RestractTower Script.

Need to be tested : https://github.com/5axes/Calibration-Shapes/releases/tag/V1.2.6

stephentt, 2021-06-03:
Hi there do you need me to test it in my environment and if so do you have any instructions on how to do so?  I have only used the plugin from the Cura marketplace before.

5axes, 2021-06-03:
Download the curapackage present in the last release on this Github ( the link to this release is in the previous post) . Drag ans drop the file un Cura to install the release. Restart Cura and test the new release . ( don't forget to updater the script with the copy script function)

stephentt, 2021-06-03:
I did as you said and it looks like the result is better.  Thank you for taking a look and responding quickly!  I got this as a result for the relevant section:
```
G91 ;Relative positioning
G1 F2700 E-3.50000
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y220 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
```

5axes, 2021-06-21:
Last release published on the marketplace. Should solved this issue.

closed 2021-06-21 by 5axes

marcoviarengo, 2021-11-27:
Hi all! 
I have a little issue. I run this plugin and I don´t get stringing with 1mm but when I am not running the plugin and I set that value to my retraction it doesn´t work. Is there something else I need to change? I've tried 20 to 40 speed, combing mode off and on.. For sure I am missing something...