5axes/Calibration-Shapes #40: Temp Tower 220-185, so what's 175 for?

jagauthier, 2021-05-24:
I installed this plugin, and created the temp tower.  I followed the guidance here:
https://github.com/5axes/Calibration-Shapes/wiki/PLA-TempTower-220---180%C2%B0C

[image]

When I inspected the gcode afterwards (to verify post-processing) I see the temperature actually goes down to 175.

```
;TYPE:CUSTOM TEMP
M104 S175
;TYPE:WALL-INNER
;MESH:PLATempTower
G1 F1800 E2362.62502
G1 F1200 X121.792 Y113.42 E2362.65192
G1 X119.917 Y113.42
;TYPE:WALL-INNER
;BRIDGE
G1 F600 X101.503 Y113.42 E2362.8356
```

Did I do something wrong?

5axes, 2021-05-24:
Your cura settings must be linked to this default value : IE Change Layer ->52  and Change Layer Offset -> 5

Cura **Layer height** = **0.16**   and check that the option **_Use Adaptative Layer_** is **not** active

jagauthier, 2021-05-24:
It was the adaptive layer!  thanks!

5axes, 2021-05-24:
Yes it's a common issue, most of the Creality user who are using the default **_Dynamic Quality_** profile doesn't notice that this profile also activated this parameter (assuming you are one of this Creality user).

[image]

closed 2021-05-24 by 5axes