5axes/Calibration-Shapes #41: Temp Towers: Gaps between steps make it unprintable

alexvonsiebenthal, 2021-05-25:
can't print the temp towers with the most recent version. The object has gaps/spaces between each temp step.
[image]

5axes, 2021-05-25:
Nozzle size ?

alexvonsiebenthal, 2021-05-25:
> Nozzle size ?

.6
But why should that fail the STL? Other STLs of similar towers work just fine.

5axes, 2021-05-25:
The design of these tower are made for a 0.4 nozzle. If you want to use them, with  a bigger nozzle  use the option 
https://github.com/5axes/Calibration-Shapes/issues/19#issuecomment-841889515

alexvonsiebenthal, 2021-05-25:
Indeed, the Remove All Holes options works. Thank you.

5axes, 2021-05-25:
Next release will automatically activate this function in such situation V1.2.5

closed 2021-05-25 by 5axes