5axes/Calibration-Shapes #42: Wiki entry on Linear/Pressure Advance Tower

dotdash32, 2021-06-07:
I was doing another calibration print, and realized there was section that I left out when initially writing the documentation.  When changing print settings, Cura changes `outer_wall_speed` to be half `print_speed`, which can cause some problems when testing linear advance on the outside of the test print.  I modified [my fork of the wiki](https://github.com/dotdash32/Calibration-Shapes/wiki/Linear-Pressure-Advance-Tower) to include a paragraph about this, and I believe it also decreases the print time, but would love independent verification on that.

5axes, 2021-06-08:
If someone could give some advice or can confirme this point, his answer would be welcome. As I'm not using Linear/Pressure Advance on my machine I cannot confirm this point myself.

dotdash32, 2021-06-08:
Unsure if I was completely clear in the OP, but here is the relevant part from [Klipper's docs:](https://www.klipper3d.org/Pressure_Advance.html)
> It can be useful to set the external perimeter speed to be the same speed as the rest of the print, but it is not a requirement.

When I did the test, it was much easier to see the effects of the test with a high speed and therefore high flow on the outer wall since that is the most visible part.