5axes/Calibration-Shapes #45: Temp Tower script settings not saved with project

JayJones65, 2021-07-01:
When opening a project in Cura that has the script enabled, it loads the values from the last time the script was used (not the last time the project was saved). This means that you need to document specific settings for each different temp tower and remember to hand-edit the script values each time.

IF there is a way in Cura to preserve the script values as part of the project, that would be VERY valuable. If not, then maybe there should be a bug against Cura to enable that option for your (and other) scripts that take parameters. 

Thanks for a great set of calibration objects and scripts! They are SUPER HANDY!

5axes, 2021-07-01:
Thanks for your comment. As it is a general question regarding postprocessing script , it should be on my point of view , managed on  the Cura level and as you say it is not specific  to my script.
On the same type of request the scripts used  are themselve not save with the project. 
But concerning your request Yes I think We could find a solution to store the values with the part.  I will think about it for a next release.

5axes, 2021-07-01:
Edit, not so easy after analyse, I keep this report open but not sure to find a solution via a standard script. I can do it via the plugin but it's in this case much more complicated.

JayJones65, 2021-07-01:
Thanks for the follow-up. Hopefully Cura will provide a better mechanism in the future for all scripts. 🙂

Take care and thanks again for a great suite of calibration objects!

Jay

5axes, 2022-05-20:
Looks like the case with Cura 5.0 , Cool I can close this request

closed 2022-05-20 by 5axes