5axes/Calibration-Shapes #46: EXtreme extrude feed at end of retraction tower

OctavioBs, 2021-07-03:
When doing a retraction tower, just in the moment the part has finished I get a vary fast and generous feed on the extruder, so fast that even the extruder stepper gets blocked (e3d Hemera).
[image]
[image]
[stringing 025 to 125-0.4-PLA T 004448-205C.txt](https://github.com/5axes/Calibration-Shapes/files/6758210/stringing.025.to.125-0.4-PLA.T.004448-205C.txt)

Excuse me if something is done wrong as this is the first time i open an issue on github.

5axes, 2021-07-03:
Do you use the last version of the script ?

OctavioBs, 2021-07-03:
Yes, 1.3.3

[image]

5axes, 2021-07-05:
This problem should not exist anymore. Did you copy the last script in the script directory ? And can you check by opening the RetractTower.py file that this is the release 1.5 ?

[image]

OctavioBs, 2021-07-09:
Now it works perfectly, excuse me for the delay but was travelling due to my job.

closed 2021-07-09 by OctavioBs