5axes/Calibration-Shapes #47: Scripts do not work when Relative Extrusion / Special Modes is activated

AndreasALu, 2021-07-05:
When the setting "Relative Extrusion" is activated in "Special Modes", the script for Retract Tower does not work.

"Relative Extrusion" puts in commands like "G1 E-0.6", so it should be easy to fix as you only need to replace the -0.6 with the current value.

One more sidenote: if you enable Retraction and enter a Retraction Distance of 0, Cura does not generate any retraction commands at all, so the scripts do not work. You have to enter a value >0. Not sure if you can show a warning from the script, but it would help finding the mistake ;-)

5axes, 2021-07-05:
Need more information , normaly it should work but the script only analyse the first extruder : 
`relative_extrusion = bool(extrud[**0**].getProperty("relative_extrusion", "value"))`

So if you use a machine with more than one extruder could be an issue ( but not sure because this settings must be common to every extruder).

AndreasALu, 2021-07-05:
I just tested it again and it seems to work. No idea what is going on. I will check further and will let you know.

And yes, I have a dual extruder machine, but it works now.

5axes, 2021-07-05:
Normaly it should work , you can have a small issue in relative mode on the first activation the previous retract distance if made with the Cura settings value , and the cancel the retract with the new value. This issue will be solved in a next release of the macro.

And Yes if you don't specify a retract value then the script doesn't work same as if you don't Enable Retractions in Cura . Perhaps I should add a test for this type of user .

5axes, 2021-11-11:
Closing the request .. No feedback from @AndreasALu

closed 2021-11-11 by 5axes