5axes/Calibration-Shapes #48: Minor doc and consistency issue

rubydesign, 2021-07-08:
But let me start with a big thanks. This is great work, put up a paypal button i'll give you a coffee.
My problem has often been that many towers and things require Simplyfied3d to slice or manual work.
And ready towers (gcode) are made for Cartesian printers, i have a delta and so they are out of the printing area.

Ok, the first doc issues i would change is: the links to say a temp tower in the readme go to the stl, but i think it would be more helpful to link to the corresponding wiki page. I can make a pr if you want

Second, the line height does not seem consistent, i have see 0.16 (temp) and 0.2 (retract) . Ideally the scripts would pick up and do the change accordingly. Failing that (too difficult/impossible) a consistent layer hight (0.2 ?) would be preferable, and mentioned in the readme.

So thanks again
Torsten

5axes, 2021-07-08:
The values for the line height are just given as indication. Some user are using 0.5 , 0.6  or 0.8 mm nozzle and idealy you should use the same  value as you have in your profil. So using just one height or showing just one height as example is not a good idea on my point of view.

Concerning the link on the Wikipage ... Yes I had the same thought this morning :)

rubydesign, 2021-07-08:
Fair enough.
But it does say on the wiki https://github.com/5axes/Calibration-Shapes/wiki/PLA-TempTower-220---190%C2%B0C#printing-conditions
to use 0.16 . And i guess for a reason.

Because i did it with 0.2 (my standard) and the layer change was not working out. So the reason being that the script default layer change works with the model size and the layer hight. But not with 0.2 or other layer hights, in which case you would have to do the maths.
Also an ok option, but maybe it should say

5axes, 2021-07-08:
0.16 it's just because it's a basic profil : 
[image]

And yes the values must be calculated according to the geometry and the layer height.  As the postprocessing script can be used on any kind of geometry and not just with the plugin models , we cannot limit the Change Values automaticaly.