5axes/Calibration-Shapes #49: SpeedTower.py uses deprecated parameter for M204

p-v-n, 2021-07-09:
According to https://marlinfw.org/docs/gcode/M204.html, M204 S<accel> has been deprecated. "P<accel>" and/or "T<accel>" should be used.

5axes, 2021-07-09:
Thanks a lot for your feedback. But I don't really knows if it's necessary to change anything for the moment. Deprecated doesn't means it's not working , the code S is still valid and some old firmware are not using the new notation.  So I will keep this issue open and waiting the feedback of more user.

p-v-n, 2021-07-09:
Sounds good, thank you!

tastyratz, 2022-02-15:
It also seems  P&T actually could create possible accuracy complications with Klipper if you're not careful.

https://github.com/Klipper3d/klipper/issues/2492