5axes/Calibration-Shapes #56: Feature Request: Multiple flow rate model in single print

rizwansarwar, 2021-08-24:
The current flow rate model only adds single model to build plate and uses the current flow rate. Although it is possible to add multiple instances of the same model and setting per model settings for various flow rates. It would be great to do this automatically just like temp tower or retraction tower for example. The advantage is that one can know good flow rate for a model in a single print. Is this possible to implement?

5axes, 2021-08-24:
Something like the SuperSlicer solution ?   
[image]

Why not but in any case as in SuperSlicer we should certainly have two prints : The first one with a range of  +/- 20%   and the second one with a smaller tolerance to fine tune the flow rate.

rizwansarwar, 2021-08-24:
Yes exactly like that. I have resorted at times to SS just for this model even though most of my workflow is in Cura. Flow rate calibration is pretty important at least in my case so saves a lot of time when you can fire one print and know ball park. Thanks for quick response, and btw your models have taken guess work out of cura at least for me so thank you very much.

5axes, 2021-08-24:
An other solution could be to use the "M221 - Set Flow Percentage" Marlin function via a Script

rizwansarwar, 2021-08-24:
Possibly, and it would probably fit in easily with the existing towers available in the plugin. My only counter argument is that I personally use flow rate testing to calibrate top surface quality (esp for ironing), I am not sure if a tower will help with there. Maybe it will but I have never tried it.

5axes, 2021-08-24:
I will make a test with this model ...
[image]

5axes, 2021-08-25:
Tested
[image]

Not very nice solution this tower
[image]

5axes, 2021-08-25:
New Function MultiFlowTest  the design of the part must be tested. Add 11 parts from 110% to 90% with the customized flow parameter already fixed to the part.

[image]

5axes, 2021-08-25:
https://github.com/5axes/Calibration-Shapes/releases/tag/V1.5.0

djcutch, 2021-09-01:
> 
> 
> New Function MultiFlowTest the design of the part must be tested. Add 11 parts from 110% to 90% with the customized flow parameter already fixed to the part.
> 
> [image]

i tested this with klipper and the flow rate does not seem to change and it printer all the layers at one vs an object per flow.

5axes, 2021-09-01:
All the layers at one vs an object per flow should not be a problem. Can you link to your post the generated G-Code for analyse ?

cptSwing, 2021-11-22:
(this is how TeachingTech aims to calibrate flow, I can't speak for how accurate it is though, https://teachingtechyt.github.io/calibration.html#flow )

5axes, 2021-11-22:
Yes it's the first basic calibration method : We have the description of this method in the Wiki : https://github.com/5axes/Calibration-Shapes/wiki/Flow-calibration
But here the request and the goal  are a little bit different it's to get in one time the correct flow , without need to make a cube , measure the wall, check if it's correct and discover that the top result is not correct even if the flow are well calculated.   I don't know if it's a good approach but it's to try to move forward on this. 
But what I can say is that TeachingTech did not understand the specificity of Cura and from my point of view, the flow calculation as described in the given link is not correct for Cura. It works for PrusaSlicer / SuperSlicer but not for Cura see my analysis in : https://github.com/5axes/Calibration-Shapes/wiki/Flow-calibration

cptSwing, 2021-11-23:
That was an interesting read, thank you!

5axes, 2023-09-05:
I think I can close this request

closed 2023-09-05 by 5axes