5axes/Calibration-Shapes #62: Display details on LCD functioning incorrectly

Gromor322, 2021-10-03:
You can replicate this with a retract tower post-processing script with the said feature enabled. It works fine when you change retraction distance but acts up when changing speed. Some seemingly random values are displayed, it can also be seen in the G-CODE. 
Here is layer 0 M117 command with retraction speed set to 5
> M117 speed (5.0/5.0)

it seems to be fine for a few first layers, but at layer 8 it is 
> M117 speed F300
which just displays F300. You can take a look at the Gcode I compiled with this script, because I can't figure it out. also attaching a pic of a message that should be "speed 10" or something.

[Untitled.txt](https://github.com/5axes/Calibration-Shapes/files/7273054/Untitled.txt
[image]
)

5axes, 2021-10-03:
It's not an issue the value display by the code is the real F value so it's a speed in mm/min   to get the value in mm/s as defined in Cura or in the script you must devide by 60  so F600 = 10 mm/s . I'm using this solution as I prefer to get the value I have to search in the G-Code if I want to analyse something.

Gromor322, 2021-10-03:
oh, got it, seems a bit inconsistent though (first layers display speed in mm/s and then it goes on to display those values in mm/min)

Thanks for your reply, perhaps consider converting those values to mm/s in the script, that would make it easier for noobs like myself.