5axes/Calibration-Shapes #65: How do you use temp tower script?

evilC, 2021-11-04:
I see the TempFanTower script in the Post Processing menu, but I don't see how to actually generate and add the model which uses these values.
Please advise

5axes, 2021-11-04:
- [https://github.com/5axes/Calibration-Shapes/wiki](https://github.com/5axes/Calibration-Shapes/wiki)
- [https://github.com/5axes/Calibration-Shapes/wiki/Customize-script#tempfantowerpy](https://github.com/5axes/Calibration-Shapes/wiki/Customize-script#tempfantowerpy)
- [https://github.com/5axes/Calibration-Shapes/wiki/PLA-TempTower-220---180%C2%B0C](https://github.com/5axes/Calibration-Shapes/wiki/PLA-TempTower-220---180%C2%B0C)

gohma231, 2022-04-03:
If I want to print the temp tower at a different layer height, I planned to use this procedure:

Desired Layer height: 0.2 mm = h_layer
Increase height of model to multiple of 0.2 mm (uncheck snap scaling): height_0 = 58.9971 mm -> height_1 = 59.2 mm
Count number of base layers: n_base = 4
Number of different temperatures: 7 = n

(h - h_layer x n_base) / (n x h_layer) = Change Layer Setting

Or for the above settings...

(59.6 mm - (0.2 mm * 4)) / (7 * 0.2) = 42 Layers per temperature change

This info is primarily for all to see or for someone more knowledgeable than I to point out potential issues.