5axes/Calibration-Shapes #70: Can't find this once installed

RDoull, 2021-11-18:
HI, 

It's probably just me but I've installed this and I can't find it. 
I expected it to be under "extentions" but it dosesn't show. I've checked all the other menus and can't see it. 
any help greatly appretiated.. 

I'm using a Windows 10 pc and just upgraded to Cura 4.12.1.  I'm installing from Cura marketplace. 

thank you for your help and time 

kind regards

Robin

5axes, 2021-11-19:
Yes it should be in the extensions menu.
Step 1 : check if the calibrationpart is present as folder in the plugins folder  ( you have a direct access via the menu Help -> Configuration folder )
Step 2 : check your log  file to analyse if you can have any issue regarding this plugin

RDoull, 2021-11-23:
Sorry for the delay in getting back. 

I've checked and the folder is there, however in the log it says 
2021-11-23 23:49:47,711 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [426]: Plugin [CalibrationShapes] has been disabled. Skip loading it.

what would I have done to disable it ? I've tried to uninstall the pugin and reinstall it again but still the same message and nothing in my extentions.. my next thought was to uninstall Cura and reinstall again fresh.. is there another way?

RDoull, 2021-11-24:
Thank you for all your help. 

I just came on to say its sorted and noticed in your last message you highlit the check box that for some reason wasn't ticked on mine..  I totally wiped Cura from my machine done a full power cycle then reinstalled it and it still didn't show.. then I installed another random plugin and in the instructions it said to tick the box... thats when I seen it and I seen that it wasn't ticked.. 
all working now though.. thank you so much for your patience with my lack of understanding.

closed 2021-11-24 by RDoull