5axes/Calibration-Shapes #76: Help with flow tower

mbeale0, 2022-01-07:
Hello! Great plugin, been using it a lot today and am now on the flow test tower. Saw someone else ask a question like this here so hope this is the right place. I read the wiki but wanted to clarify what I am looking for. Is it for the pin to slide in and out with no resistance, or to have some resistance but easily come in and out or to not come out but easily go in? I have 92-96 about the same , going in and out but but wanted to make sure those would be the correct ones because I [previously thought I had under extrusion issues so didnt expect a lower flow rate to work better.

Thank you! and keep up the good wokr

5axes, 2022-01-07:
That's a very good question ! To be honest I'm Not very satisfied of this test. In the Online version the difference between the pin Width and the female geometry is just 0.05mm : **wo=w+0.05**

For Cura that's not enough, but this value should be dependant of the Layer Height parameter, so on a theoretical point of view it's hard to define a fixed geometry to define what is the best flow parameter whithout fixing a layer height value.  What's why I think this test is not a good one.

At the present time, I'm using something like  : 
```
lay=0.2; // Layer Height
w=4; // Pin Width
wo=w+lay; // Housing Width
```

But I should make some test to find the best compromise. But as it was not in my top priority if you have time to work on this point so if you have times to work on it, your feedback on the issue would be greatly appreciated.

mbeale0, 2022-01-07:
I'm pretty new to all the calibrating things for 3d printing so I may not be thinking of this right, and probably dont have much to offer, but could you measue the depth of  pin instead? something that should be consistent no matter the layer height? or maybe that isnt as affected by changing the flow

tkang2020, 2022-12-04:
> 

Have you tried the Multi-Flow test part within the Cura Parts for Calibration extension? I tried that one and it gave me a better idea of how the first layer adhesion was affected by flow, as well as the top layers since it's about 4-5 layers tall per card. Gotta have a level bed to make sure it's not a factor, but they're very small and quick to print. Github Doc also has some decent examples of over/under extrusions in pics.