5axes/Calibration-Shapes #78: Temperature Tower Walls Too Thin?

PICKLEBOY, 2022-01-12:
I am attempting to run the temperature tower for PLA+ and Cura is not slicing the entire model (Cura Version 4.12.1). From what I can tell it has to do with how thin the walls are on the Temperature tower? I am using a 0.4mm nozzle and have my line width set to 0.4 and this is what Cura does when I try to slice it. It does this with all of the temp towers. If I change the line width to 0.1 all of it will slice, but I do not have a nozzle that size.
[image]

I have been following this tutorial https://www.youtube.com/watch?v=EAG_538k_bM.

5axes, 2022-01-13:
Well Chep Sometimes is not good for everything.  Set the Outer Wall Inset parameter to 0 as it should be by default And it will be ok

----- Mail original -----

tqrj7ibubbb81

PICKLEBOY, 2022-01-13:
Thank you for that! It will slice now.

closed 2022-01-13 by PICKLEBOY