5axes/Calibration-Shapes #79: DimensionalAccuracyTest - Imported model doesn't match wiki width

kusold, 2022-01-19:
When you import the model with Cura, the bottom width is 25mm and the top width is 2.5. The wiki indicates that the models should go from 30 - 5.

[image]

5axes, 2022-01-19:
Yes, you are right the Wiki is linked to the current models on this Github and if you use the Package avalaible on the Ultimaker marketplace , the model is the previous one. I hop to update the release on the Ultimaker marketplace soon.

5axes, 2022-01-19:
Ultimaker usualy takes 15 days to validate a plugin ( sometimes more if they are in a hury to release a new cura version). In the meantime you can find a version on the github here : https://github.com/5axes/Calibration-Shapes/releases/tag/V1.7.0

kusold, 2022-01-20:
Thanks for getting back to me on this. It's good to know that I just caught this in between release cycles and that it'll be automatically resolved soon.

Thank you for your work on this plugin. It's easily the easiest way to get things dialed in.

5axes, 2022-02-09:
Problem normaly solved with the last version on the marketplace

closed 2022-02-09 by 5axes