5axes/Calibration-Shapes #85: SpeedTower script drop down command for "speed" is not showing up in Cura

tastyratz, 2022-02-15:
[image]

I am running Cura 4.13.1 with latest extension. The SpeedTower.py script appears to be missing the option for speed.
I wanted to run a speed test with this model https://www.thingiverse.com/thing:3249901
But when I went to add the script it appears that I am unable to select speed as an option in the script.

5axes, 2022-02-15:
It's normal, The Script and the modification on Github are up to date, but the package on the Ultimaker market place is an "old release" . So if you want to use this option, you must gat the script rirectly from this repository.

tastyratz, 2022-02-16:
Excellent,
I did grab them. That might not be a bad suggestion for the github readme file to specifically call out how to install the latest updates from github outside of the marketplace by copying to the
 %appdata%\cura\(Versionnumber)\plugins\CalibrationShapes \CalibrationShapes\ 
folder

tastyratz, 2022-02-16:
Also, another note for the documentation:

M220 Speed is a percentage, not an absolute number. If that's what is being used then the print needs to be sliced at 100mm/s to test absolute speed numbers when printing. Don't forget other speeds, like infill, walls, initial layers