5axes/Calibration-Shapes #89: Temp tower script not working

JaVelin-X, 2022-02-19:
In Cura 4.12 I am trying to print a temp tower but I noticed the script wasn't changing the temperature as it was printing. I checked it by searching for M104 and they are not being added that I can find. I see the temp is not changing on the printers screen as it prints

5axes, 2022-02-19:
Pebcak

JaVelin-X, 2022-02-27:
Works on one PC but not this one...
[image]
[image]

5axes, 2022-02-27:
Don"t know what is this big 3d format . But the script are just for text and Marlin format.

JaVelin-X, 2022-02-27:
the second PC .. i'll try with regular Cura format on the sho computer too if you think it makes a difference. BTT one jjsy embeds a thumbnail of the model.

[image]

JaVelin-X, 2022-03-01:
yes with all other add-ons disabled. the temperature scrip doesn't work on this PC but the changeAtZ script does

5axes, 2022-03-02:
Sorry absolutly no idea, you will be oblige to investigate by yourself.

e-SonOfAnder, 2022-03-15:
I am experiencing this same issue - Cura 4.13.1, add temp tower, add script, and after slicing the resulting gcode does not contain any of the M104 commands to change the temperature during the print. The retraction distance calibration item worked, but I am unconvinced the retraction speed one worked either, as the full print came out with no issues, but I have not looked at that gcode yet. The Calibration Shapes is the only extension I have installed from the Cura marketplace.

If there are logs or any additional diagnostic information I can provide, please let me know as I'd be happy to do so.

5axes, 2022-03-15:
Screenshot of your script settings ?

e-SonOfAnder, 2022-03-15:
Here you go:

https://imgur.com/a/9nU5mqR

e-SonOfAnder, 2022-03-15:
Just ran it again this morning, and it seems to have worked; this may be a Cura bug requiring it be closed completely between projects (similar to the current open one that causes Cura to be unusable if you tell it to open a gcode file until you close and restart it).

5axes, 2022-03-15:
I don't know, Everything seem's to be correct in your settings. Should be ok directly without need to restart.

DarokCx, 2022-03-28:
Similar problem here. Using a .6mm nozzle with .4mm layer height. set tempfan tower to change at each 20 layer, but in reality it does every 40 layers or so. Seems it looks for a certain z height instead of layer number.

jvolker, 2022-05-20:
Same in Cura 5.0.0

5axes, 2022-05-20:
tested again right now in Cura 5.0.0 works fine for me.
[image]

jvolker, 2022-05-20:
I'm sorry, I just realized I first didn't use any post-processing script and I was then not using the right script. It seems to work now. Could be good to give a hint after adding a temperature tower or activate the post processing script automatically.

Thanks.

5axes, 2022-05-20:
@5axes how did you get the color scheme to represent temperatures? -> I'm Using a GCode Viewer  ( SuperSlicer GCode Viewer) 
https://github.com/supermerill/SuperSlicer

5axes, 2022-06-29:
I think I can close this issue now

closed 2022-06-29 by 5axes