5axes/Calibration-Shapes #91: Feature request: write text

ImagineerNL, 2022-03-04:
Might not be a specific calibration thing, but would be used for calibrating different stls.

Usecase; i have multiple versions of the same item; for example scaled 99 / 100 / 101 percent for making a tight fit.
However, on the print plate it can be hard to see the difference between them. When it is using a different infill it's even harder since the differences arent visual.

It would help a lot to be able to add 1layer thick text-meshes, so i can 'annotate' on the buildplate next to the mesh and be able to see it next to the finished print a couple of hours later. 
No need for special things as font etc, just a solid straight font like arial bold will do.

5axes, 2022-03-05:
Nice idea,  Perhaps not a free text filed but just a mark
[image]

ImagineerNL, 2022-03-05:
That would be perfect already, awesome

5axes, 2022-03-05:
ok adding for the next release. you need to select each geometry. the selection order is important because the number will be fixed according to the selection order.
[image]

5axes, 2022-03-05:
https://github.com/5axes/Calibration-Shapes/releases/tag/V1.9.0

ImagineerNL, 2022-03-05:
Thank you very much!

closed 2022-03-05 by ImagineerNL

5axes, 2022-06-13:
Function removed from the Calibration Plugin and integrated into a new Plugin dedicate to this type of function : [NameIt](https://github.com/5axes/NameIt)

ImagineerNL, 2022-06-13:
You're awesome!