5axes/Calibration-Shapes #92: Acceleration Tower script using deprecated M204 syntax

e-SonOfAnder, 2022-03-15:
Noticed that the Acceleration Tower test did not show any changes to the print at the layer changes, and found that the script is adding the following lines:

M204 S400
M204 S450
etc.

Checking the Marlin Wiki, this syntax is deprecated, and is likely why this is not having any effect: https://marlinfw.org/docs/gcode/M204.html

5axes, 2022-03-15:
Deprecated doesn't mean it's not working but just there is a new way to write this instruction. And M204 S is still working, but you should not activate the Acceleration management in Cura otherwise Cura will write is own M204 line and the instruction added by the script will be not operant.
Note that Cura itself is using the format M204 S to manage the acceleration in the Gcode File :
[image]

p-v-n, 2022-06-29:
Possibly a dupe of https://github.com/5axes/Calibration-Shapes/issues/49

5axes, 2022-06-29:
Yes it's the same discussion . And same conclusion for the moment.

closed 2022-06-29 by 5axes