5axes/Calibration-Shapes #96: Retraction Tower Part Has Bridges That Seem Unnecessary For Tuning Retraction Speed and Distance

borland1, 2022-05-02:
Tuning my printer with PETG filament.  I have stringing under control, but the unsupported bridging in the Retraction Tower Part is causing print failures.  Nothing said in the documentation about what the bridges are there for in tuning the retraction settings.

I don't have any part fan control as my extruder's fan duct only diverts part of the flow to the part.  As a workaround, I should have added supports under the bridges.

Suggest part bridges have ability to be removed by a script option (check mark) or by modifying the part to remove the bridges.

5axes, 2022-05-03:
Any suggestions for a more appropriate design are welcome. It is not possible to remove the bridges using the scripts but another design is possible.
The presence of the bridges is there to make a clear separation between the zones and normally make a kind of reset of the stringing before starting the new zone.  The objective is also to make the print not monotonous to avoid that the stringing creates zoness of accumulation of material. 

But I agree with you that this model may not be the most relevant and that we could perhaps gain in printing time.

borland1, 2022-05-03:
Well the part seems fine with the exception of the bridging.  There are already numbers on one side of each level and detents between levels, so the bridging really doesn't add much.

My fan duct is less effective at cooling the bridge as the level rises higher above the bed.

[image]

ImagineerNL, 2022-06-20:
@5axes commenting out line 74 (under the //liason) in the SCAD file removes the bridge.
Other retraction tests like e.g. TeachingTech indeed also dont have the bridge in it and it might create interpretation errors when bridging errors obfuscate retraction tests.

p-v-n, 2022-06-29:
I agree, bridges are unnecessary. Here is a model I just made in a few minutes.
Shared under CC BY-NC-SA 4.0 license (https://creativecommons.org/licenses/by-nc-sa/4.0/)

`layer_height = 0.2; // [0.1 : 0.01 : 0.3]
nozzle_diameter = 0.4; // [0.1: 0.1: 1.0]
pole_radius = 5; // [1 : 0.1 : 10]
poles_distance = 50; // [5 : 0.1 : 100]
$fn = 30; // [10 : 1 : 100]

test_height = 500 * layer_height;
wall_width = 2 * nozzle_diameter;
base_brim = 10 * nozzle_diameter;

base_width = 2 * (base_brim + pole_radius) + poles_distance;
base_depth = 2 * (base_brim + pole_radius);
base_height = 4 * layer_height;

cube([base_width, base_depth, base_height], center = true);

difference() {
    union() {
        translate([-poles_distance / 2, 0, base_height / 2]) {
            cylinder(h = test_height, r = pole_radius);
        };

        translate([poles_distance / 2, 0, base_height / 2]) {
            cylinder(h = test_height, r = pole_radius);
        };
    };

    translate([-poles_distance / 2, 0, base_height / 2]) {
        cylinder(h = test_height + layer_height, r = pole_radius - 2 * wall_width);
    };

    translate([poles_distance / 2, 0, base_height / 2]) {
        cylinder(h = test_height + layer_height, r = pole_radius - 2 * wall_width);
    };
};`

I setup a 0.1mm retraction increase every 5 layers, starting from layer 4. That way I virtually continuously tested all retractions between 0mm and 100mm.
@5axes , if you decide to use the above in this project, feel free to modify it as you see fit.

5axes, 2022-06-29:
On my point of view no added value, no identification of the zones, how to determine the value that is acceptable and why test every 0.1 mm? 
[image]

ImagineerNL, 2022-06-29:
Then i'd suggest using my solution, which uses exactly the same model, just without the bridges

https://github.com/5axes/Calibration-Shapes/issues/96#issuecomment-1160291841

p-v-n, 2022-06-29:
> On my point of view no added value, no identification of the zones, how to determine the value that is acceptable and why test every 0.1 mm? [image]

My apologies for not mentioning how to derive the acceptable values as I mistakenly though that it's obvious and doesn't need explanation.
To derive the acceptable value, one should identify the *minimum* height `H`, where there the stinging is the smallest. Then, assuming that the starting value at height `0` (top of the base) is `R0`, and the end value at `Hmax` is `R1`, then desired value can be calculated using this formula:

`R0 + (R1 - R0) * H / Hmax`

For example, if the starting value is 1, the end value at height 100mm is 11, and the minimum stringing is observed at height 40mm, then the acceptable value is `1 + (11 - 1) * 40 / 100 = 5`.
This is similar, for example, how Klipper suggests doing pressure advance calibration: https://www.klipper3d.org/Pressure_Advance.html.

As to why 0.1mm, it just gives a gradual transition. One can choose bigger steps, i.e. more coarse granularity, and, perhaps, shorter model.