5axes/Calibration-Shapes #99: PLA Temp Tower 220-180

Willemthefirst, 2022-05-31:
Hello,
I tryed the Temp Tower twice with different Filament. On the Backside right hand inner corner there are always two little line to long.
See Picture.
[image]
Willemthefirst

5axes, 2022-06-01:
Cura 4.x or Cura 5.0 ?

5axes, 2022-06-01:
Could be an issue in this case with the parameter :  "Wall Ordering"   the default value brings a lot of problem. Try to set this value to "Inside To Outside".

Willemthefirst, 2022-06-01:
Thanks, I'll check it out.

closed 2022-06-02 by Willemthefirst

Willemthefirst, 2022-10-11:
Hello, in my case it was the check on "Optimize Wall Printing Order".
Unchecked that and the corner is ok now.
So thanks again for the fast answer

Kurt