5axes/Calibration-Shapes #100: Retraction Tower Script Not Working

smartalx, 2022-06-08:
Maybe I'm not understanding how this post processing gcode is supposed to work, but I'm sure I followed the instructions on the wiki.  As I understand it a retraction is a G1 code with a negative E value.  I'm checking the produced g-code and I don't see any E- codes produced for the retraction tower, except for the start and end of the file.  Plus when I slice the model without the post processing gcode, there's no change in the print time and no change in the retraction time compared to when I slice the model with the post processing gcode.  I'm pretty sure I'm doing everything right.

Now I am using a 0.6 nozzle but I put a check in the "Remove all holes" parameter as instructed.
Cura 4.13.1

[image]
[image]

5axes, 2022-06-08:
The name of these scripts are "Postprocessing" script. This means that they are activated at the time of postprocessing the Gcode, i.e. at the time of writing the final code. So there is no modification of the informations in Cura. The only way to see if the script works is to compare two Gcode files or to reload the Gcode in a Gcode Viewer.
[image]

5axes, 2022-06-08:
**Additional note :** 
The retraction **must be activated** in the Cura Project. If you don't have any E code in your file, that's the origine of the issue.
In the Gcode if you have activated the option "Display Detail on LCD" You should find some **M117** instructions in the GCode.

[image]

smartalx, 2022-06-08:
I did look at the gcode.  And retractions is turned on.

smartalx, 2022-06-08:
There are M117 instructions in the code.  But there are only 3 lines with "E-", which as I understand it is the retraction.  E for extruder.  Negative number means retract instead of extrude.  The value will be how many millimeters to retract.  I see 1 example at the top of the gcode before the print starts and 2 at the bottom after the print is done.

5axes, 2022-06-08:
Vous devez partager votre projet pour l'analyser. Sans exemple impossible d'analyser votre problème.

smartalx, 2022-06-09:
I don't understand.  My project is the retraction tower.  Do you mean the gcode?

closed 2022-06-09 by 5axes

smartalx, 2022-06-09:
dammit
no
I must have uploaded the wrong files.  I switched to the retraction test because I couldn't wait 3 days to run the calibration tower.

smartalx, 2022-06-09:
The 3mf file doesn't create retractions.

smartalx, 2022-06-09:
But that's just a smaller extrusion.  Is a smaller extrusion the same thing as a retraction?  I thought a retraction was a gcode like:
G1 F2700 E-1
And there are only three of those "E-" occurrences in the file...
None of them during the printing of the layers.

The reason I even noticed this is the retraction towers I printed don't change at each level.  There's no visible difference between the stringing at the lower level and the higher level.  But I'm experiencing retraction problems with my normal prints.  So I need a retraction tower that shows me the problems at 1mm up to 6mm or up to 10mm so I can figure out which one works for my printer.  And the calibration tower doesn't work for that.  All of the towers look identical, whether the retraction is 1 or even if it's as high as 10!

5axes, 2022-06-09:
I don't doubt that you have a retraction and stringing problem. But it's a problem between your machine and your filament. Not a script problem in this case. 

On my point of view you should first of all have a look to the parameter on the maximum Extruder speed on Ender-3 . 25 mm/s is not enough try to increase to 50mm/s

hayden-t, 2022-07-02:
I was wondering about this too, as my tower seemed fine on all layers, i tried looking for the E- commands too and could not find them, but did find the lcd lines.  thanks

5axes, 2022-07-03:
If you have the lcd lines. Then the script is working. Stringing is not an issue for everyone. Sometimes just with the minimal value you don't have any issue so with or without change the result looks correct on every section. The first step is to tune your printing temperature via the temptower. A low temperature could produce a nice visual part, low stringing but could generate a poor adhesion   between layers. Definin
g the right profile is sometimes a long process, that's why the plugin is offering so much test parts