5axes/Calibration-Shapes #101: FR: Squish Square

PhilBaz, 2022-06-18:
I've started including 2 30mm x 30mm x .31mm squares in every print. I use 'layer start' X & Y to ensure they are printed first. I use these to judge and adjust first layer squish. Better than starting the print over.

I have an enclosed printer and do lots of ABS. With thermal expansion constant revaluation is require for back to back prints....unfortunately.

Would love to have these shapes included in your plug-in.  I think its very useful. I currently import them for every print, as its faster than resizing a cube.

[image]

ps....have you seen the stuff DALL E 2 is making....Bananas! 
[image]

5axes, 2022-06-18:
Hello,

Why a square and not a round part ?  And why don't you use the skirt as feature to adjust your layer height as everyone is doing ?

PhilBaz, 2022-06-18:
@5axes 

It could be round,,, Though I think objects with corners tell you more about it being squished too much... I would have to do a skirt with 10 laps to see the info these squares give me. Also, I need to see both walls and skin. I do one square as concentric, that is where you can really tell if you are too close. You'll see problems on concentric you wont see with lines/zigzag skin.

I do mostly ABS, that requires more squish, and at times a brim......the margin for squish can be much smaller with ABS. Often tipping on the side of too close. That messes up the brim creating a wavy edge, that further complicates removing the brim, and leaves an ugly edge on the part.