5axes/Calibration-Shapes #102: Linear Advance & Acceleration & Ringing Tower

traviselam, 2022-06-20:
Just wondering if the readme for the Acceleration & Ringing Tower should mention disabling linear advance in Marlin firmware. 
The link in the  readme (https://www.klipper3d.org/Resonance_Compensation.html) mentions disabling pressure advance. This is called "Linear Advance" in Marlin.

5axes, 2022-06-21:
Hello,

As I don't have this feature on the Marlin of my machine (Ender-3) I cannot have an opininion on your request. Other advice will be welcome.

traviselam, 2022-06-21:
Gotcha. Yeah, I just updated to the latest Marlin firmware on my machine (Ender 3) and I've enabled linear advance when compiling the firmware. I just wanted to make you aware of it just in case. Thank you so much for the excellent plugin! The shapes and tests have helped me so much already!

closed 2022-06-21 by traviselam