5axes/Calibration-Shapes #103: SpeedTower script leaves feed rate at low percentage on completion

tillig, 2022-07-30:
When running the SpeedTower.py script using the "speed" option, it seems the speed is manipulated [using `M220`](https://github.com/5axes/Calibration-Shapes/blob/654d5b5ea3a208d40b12a764f7fd6348ff012729/resources/scripts/SpeedTower.py#L132) which sets the feed rate of the printer.

Unfortunately, after the slice and print of a speed tower, it _leaves_ the feed rate at the altered percentage rather than returning it to 100%. On Ender 3, that's a persistent setting that isn't reset automatically on the next print. If you're not paying attention, all subsequent prints you run will be at the reduced feed rate until you manually set it back to 100%.

I think an `M220 S100` at the end of the script would be a nice addition that would potentially have saved me from having to rerun a bunch of tests due to the odd feed rate being left behind.

tillig, 2022-07-30:
[It seems there's a backup/restore command for feed rate](https://www.3dprintbeast.com/marlin-m221/) so for speed, I gather doing `M220 B` at the start of the script (to backup the current setting), then running the script, then `M220 R` to restore what the user had set before would maybe be even better.

5axes, 2022-08-04:
M220 B and M220 R looks like a good idea . I will try to add this option in the next release of the script

5axes, 2022-08-04:
Ok added on the Github .. Must be now tested on my printer.

5axes, 2022-08-11:
Tested on Ender-3 Doesn't work ..  M220 B and M220 R  doesn't Restore the speed percentage on my Machine. But on other hand doesn't produce any issue. So I will leave it like that. If someone could be successful leave a message here.   
On other point the feed percentage is reset to 100% if you switch off / On the printer so it's not a big issue for a normal use. Just need to take care during the test .

tillig, 2022-08-11:
Maybe something to note in the docs, then. I don't switch my printer on/off between prints/tests so maybe that was part of my gotcha. It just was unexpected.

5axes, 2023-05-24:
It might make sense to mention it in the Wiki. But at the moment, as I haven't had any feedback on how this code works and on the other hand the presence of the code hasn't caused any particular issue, I don't see the needs of mentioning it in the Wiki.