5axes/Calibration-Shapes #104: SpeedTower 'speed' option assumes initial wall speed of 100mm/s

tillig, 2022-07-30:
When using the "speed" option for the SpeedTower script, the "starting value" is [actually a feed rate percentage](https://github.com/5axes/Calibration-Shapes/blob/654d5b5ea3a208d40b12a764f7fd6348ff012729/resources/scripts/SpeedTower.py#L132), not a number in mm/s like speed. While that's noted [in the details about the script proper](https://github.com/5axes/Calibration-Shapes/wiki/Postprocessing-scripts#speedtowerpy), it's not mentioned on the [Acceleration Tower wiki page](https://github.com/5axes/Calibration-Shapes/wiki/Acceleration-&-Ringing-Tower) where the script is used. On [the Acceleration Tower page](https://github.com/5axes/Calibration-Shapes/wiki/Acceleration-&-Ringing-Tower#printing-conditions) it says the printing conditions are:

> Wall Speed : 80 mm/s (Use sufficiently high speed, around 80-100 mm/s, for external perimeter)

To work with a percentage-as-mm/s, it actually _needs to be 100 mm/s_.

Granted, there's really not much actual wiki docs on the Acceleration Tower page about the _speed_ aspect of the script, so much as there is acceleration and jerk, but I found all this out by accident and it cost me a lot of time actually getting a speed calibration done correctly. It might be good to enhance the docs on the Acceleration Tower page to include a bit more about speed, and probably mention the wall speed should be 100 but only if you're doing the speed test.

5axes, 2022-08-04:
All this seems normal because the so-called "Acceleration & Ringing Tower" test is normally done to deal with acceleration and jerkiness.  That's why, even though the script allows this setting, there is no mention of the possibility to play with the speed.

tillig, 2022-08-04:
I just thought it might help if the default documentation just said "set it to 100" instead of saying "set it to 80, but any value 80-100 would work" (basically). It's very much not clear without a lot of manual putting-the-docs-together-in-your-head that the same test tower, using the same script, really _can't_ be run with 80 to get a valid result.

Or maybe a speed section on that wiki page for the Acceleration Tower would help clarify, so it could have its own settings section to explicitly show it needs to be 100? I just know I spent a lot of time trying to figure this out and understand why the results I was getting weren't quite right, and it was because I kept using 80. I was thinking some clarification in the docs could save other people the time I lost.

5axes, 2022-08-04:
Did you also play with the : 
Junction Deviation , Marlin Linear and RepRap Pressure option ?

tillig, 2022-08-04:
I haven't messed with those as yet. I only get maybe an hour or two every couple of days to work with my printer (basic Ender 3, first version) and after updating it a _bunch_ recently (direct drive, all-metal hotend, linear bearings for X axis, silent fans) I've been just trying to get the simpler stuff dialed in - temperature, retraction, speed.

5axes, 2022-08-11:
Additional information : Speed tower script doesn't assumes initial wall speed of 100mm/s   the speed option is using the M220 instruction so it's like if you set on your printer an over-speed value. Up to you to calculate the correct speed by using the appropriated factor. Personnaly I'm using this script most of the time with the **Speed Option**  with a starting Value of 100 (**100%**) and a incremental value of 5 or 10 (**5% or 10%**) 
[image]