5axes/Calibration-Shapes #105: TempFanTower script 'fan' option not compatible with bridge settings

tillig, 2022-08-01:
When enabling 'fan tower' option on the TempFanTower script, if you have bridge settings enabled (which is noted as being recommended in the docs), the bridge cooling fan settings take over when printing the bridge part and then it reverts to the fan tower settings for the left/right columns.

It's unclear if that's the desired effect, but it does make testing bridge fan settings difficult because it means you can only print one tower test at a time (or use the spiral bridge test) to see the differences in fan speed for bridges.

5axes, 2022-08-04:
Ok this point need further analyse.   I don't know for the moment if I need to overwrite the Bridge Fan instruction in such case or keep the Bridge value or even use this values only in the Bridge situation to add a new test.   This point still needs to be validated.

5axes, 2022-08-05:
Script updated with new option : 
[image]

Even if it's again an option in the script and it's not indicate as bo be used , it's good to have the choice in case of .

closed 2022-08-05 by 5axes