5axes/Calibration-Shapes #106: Accel tower doesn't slice

GregValiant, 2022-08-03:
The Accel tower doesn't slice when brought into Cura 5.1.  It did slice after being rotated 90°.
The repair report indicated:
--> 18 Degenerate faces (?)
-> Vertex count changed from 13504 to 12747 (-757)
-> Triangle count changed from 27022 to 25504 (-1518)
Here is the link to the [UMCura bug](https://github.com/Ultimaker/Cura/issues/12932) report.
Here is a repaired model if you need it (or even if you don't).
[GV_accel_tower_fixed.zip](https://github.com/5axes/Calibration-Shapes/files/9252575/GV_accel_tower_fixed.zip)

5axes, 2022-08-04:
As usual without project  (*.3mf file) I cannot investigate this issue.  On my Side I haven't any issue to slice this model and the MeshTool Plugins or Cura itself doesn't report any problem concerning this model.
[image]

5axes, 2022-08-04:
Issue comming from the X size of the build plate ????   220 should be the new number of the beast . Don't ask ne why .

5axes, 2022-08-04:
Solved  (Temporary solved) by changing the external size of the model 121x121 , but it's still a Cura weird bug not totaly solved.

GregValiant, 2022-08-04:
This appeared in my cura.log file after a failed slice:
Backend exited abnormally with return code 3221225477!
You can close this if you like.
I spent some more time on this but all I've come up with is the same as you..."It's the build plates fault".  On the surface that shouldn't be possible.  Moving the model anywhere in the X other than the load position also allowed it to slice.  Moving it back to the load position (0,-25,0) and it sliced.
That's enough.

TheHolyRoger, 2022-08-05:
Thanks, moving the model also fixed it for me

GregValiant, 2023-04-23:
@5axes The OP on Cura 15328 updated to Cura 5.3 and the problem went away.  I've closed that one.