5axes/Calibration-Shapes #113: Post processing scripts not working

g4njawizard, 2022-09-07:
Hello everyone,

I am still calibrating my Ender 3 S1 Pro and I just realized that the Post Processing Scripts are not working.
I use Cura 4.13.1
I also have the newest Ender 3 S1 Pro firmware installed.

I just tried to print the PLA+ Temp Tower with post processing script active and I saw that the temperatures didn't go down. Thats why also all my retraction tests looked on all stories the same. How to verify that the gcode has been added with the script after slicing?

For example:

I added the script with

starting at 230°

temperature increment -5

change layer offset 5

Cura's esun PLA+ print set temperature: 205°

IMO it should have startet with temp at 230 beginning from layer 6 right? it didn't ..

Is there a unlocked/professional Firmware for the S1 Pro that might help with that? I can't use the newest Cura version. Tried it and all prints and went crazy, including the nozzle to move weirdly...

5axes, 2022-09-07:
Can you see in the Gcode  the modification ( Ie M117 comment) .

Please share your Gcode File

g4njawizard, 2022-09-07:
Hi, no M117 line found.
Cant upload here or to pastebin or similar due to it's size or file type.

g4njawizard, 2022-09-07:
[Untitled.gcode.zip](https://github.com/5axes/Calibration-Shapes/files/9505794/Untitled.gcode.zip)

5axes, 2022-09-07:
Your Gcode is correct , we can see that the temperature is modified at different layer ( Even if your settings **Change Layer** is not correct certainly because you are using 52 for a layer height of 0.2, the default value of 52 is for a layer height of 0.16  see : https://github.com/5axes/Calibration-Shapes/wiki/Customize-script )

[image]

If you have an issue it's on the printer side not on the script, but on this point I cannot help you I'm not an user of this machine.

closed 2022-09-07 by 5axes