5axes/Calibration-Shapes #115: Retract Tower distance test fails

pierremilcent, 2022-09-13:
I built a retratct tower project to validate which distance best fits the printer (Sidewinder X2).
The printer is a direct drive, with BLTouch.

I don't understand why the extruder motor is running so fast and long, while i only configured 0.25mm retraction at first with 0.25 increment. default speed is 25mm/s.

I aleady build a Retract Test with the calibration plugin (which is updated to the latest version and it works fine, as a temp tower.

Could you have a look at this 1mn video https://youtu.be/I-RQWQdmti4 

I also attach the gcode file if needed. Thanks for your support.

[RetractTower Cura 5.1.gcode.zip](https://github.com/5axes/Calibration-Shapes/files/9558195/RetractTower.Cura.5.1.gcode.zip)

5axes, 2022-09-13:
Yes there is a problem
[image]

Version of the plugin ?

5axes, 2022-09-13:
Ok I got it ...
[image]

Someone I don't know who found the good idea to add the Code M83 (relative extrusion) but commented in the Start Gcode. Unfortunately the Script detects this code and switches the extrusion to Relative mode.   

[image]

Solution N°1 remove this line temporarily
Solution N°2 modify the script in a next version

pierremilcent, 2022-09-13:
By the way i see you are using prusaslicer to view the gcode and you are able to see extruder activities : yellow and orange lines i guess ...
Can you share which option in prusaslicer are you using for such purpose ? Thanks.

5axes, 2022-09-13:
I don't use PrusaSlicer but SuperSlicer .. It's much more advanced compare to PrusaSlicer

pierremilcent, 2022-09-13:
I am also trying this version despite it is not well designed for my Mac ... and how to view these feature in SuperSlicer ?

5axes, 2022-09-13:
Use the Superslicer-Gcodeviewer , Type of functionnality  and  then in option you can also specify Retractions / Deretractions etc etc  You need to play qith the different options.

pierremilcent, 2022-09-13:
Thanks. You're right SuperSlicer does show more than PrusaSlicer. But it 

PrusaSlicer
[image]

SuperSlicer
[image]

Here i can see on the left that E value is far too large ...
[image]

5axes, 2022-09-13:
**PrusaSlicer** 

[image]

**SuperSlicer** 

[image]

+ not visible Feature like G02/G03 in SuperSlicer not in Prusa  and other functions but I forgot the exact list

closed 2022-09-13 by 5axes

pierremilcent, 2022-09-13:
Working. Solved. Thks

[image]