5axes/Calibration-Shapes #118: Temp Tower only printing base

Fishhooks1945, 2022-10-17:
Cura 5.1.1
Dremel 3D45 (with [printer plugin](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin) to make .g3drem file for my machine)
PopOS! 22.04

layer height .2mm
52 and 5 for layer and layer offset. I know this is wrong for my layer height. I was using [AutoTowers plugin](https://github.com/kartchnb/AutoTowersGenerator) and it was was only printing the base, so I went to the source and tried here. 

I am attempting to print a PLA temp tower. It finishes printing the base, and then says "Build Complete" and finishes.

5axes, 2022-10-17:
Could be an issue with the dremel plugin if you have the same issue with the postprocessing temptower script and the autotower plugin

Fishhooks1945, 2022-10-17:
def an issue with my STUPID POS proprietary printer and its dumb proprietary format. Sorry for wasting your time, I hope to use your awesome plugin eventually once I get this other issue sorted out

Don't buy dremel 3d printers, unless you like being fustrated

closed 2022-10-17 by Fishhooks1945