5axes/Calibration-Shapes #119: M104 vs M109 in temp tower script

Fishhooks1945, 2022-10-19:
Hello, I noted that the post processing script for the temp tower uses M104 Sxxx for temp changes. M104 does not wait for the extruder to meet it's target before moving to the next line, whereas M109 will wait. 

Wouldn't it make sense to use M109 and wait for extruder to be at correct temp before printing a given section?

Thank You for your time making this sweet plugin :)

5axes, 2022-10-19:
In a temptower usualy we start from the higher temperature to the lower and so the changing time in this case as the machine is extruding material is very short. By the time you are printing the inner contour of the first layer you will have reach the new temperature. So you will not see the difference on the part.

 

Using a M109 S code to wait the right temperature is also useless. This code are waiting that the nozzle have reach the programmed value. But as you are always at a temperature higher than the new target value, the machine doesn't wait anything and using M104 S or M109  S doesn't change anything in a temptower print style program.  To wait the decreasing target you must use the M109 R   code.

https://marlinfw.org/docs/gcode/M109.html

 

But this is not the choice I have made in my script because user must be able to fix a negativ or positiv increment (even if on a pratical point of view it's always better to decrease the temperature)

Fishhooks1945, 2022-10-19:
Thank You for the explination!

closed 2022-10-19 by Fishhooks1945