5axes/Calibration-Shapes #120: Retraction tower script not processing

misterpok, 2022-10-22:
Hey, first of all, thanks so much for this plugin and all your work maintaining!

I've done some towers using your plugin and script with good results.

However the retraction tower script does seem to be altering the gcode at all.

I'm getting no M117 codes, and every layer has a decrease of the cura setting in the E absolute value.

Steps to reproduce
Place retract tower from extension menu.
Adjust retracttower post processing script to distance, value 1, increment 1, layer 38, offset 5, check display details.
Slice

[image]

[retract-tower-not-post-processing.txt](https://github.com/5axes/Calibration-Shapes/files/9845087/retract-tower-not-post-processing.txt)

5axes, 2022-10-23:
The script parameter values are correct , could be an issue with your thumbnail script. Try to use the retract tower first then the thumbnail script

5axes, 2022-10-24:
Will be solved in a futur release ( Script Version 1.10)

closed 2022-10-24 by 5axes