5axes/Calibration-Shapes #122: Retract Tower: include M203 to set max E speed?

mhaas, 2022-11-01:
The wiki page for the retract tower has a [note](https://github.com/5axes/Calibration-Shapes/wiki/Retract-Tower#note-1) stating that the max speed on the ender 3 is by default set to 25mm/s and recommends to increase this to 50mm/s.

I'm pretty new to all this, so consider this a humble suggestion: would it make sense to include [M203](https://marlinfw.org/docs/gcode/M203.html) in the post-processed gcode to increase the limit to the maximum configured for the post processing script?

5axes, 2022-11-01:
Thanks a lot for the idea.  But it 's difficult to integrate  : 
A) as we cannot test the actual value on the printer, the maximum value could be already higher than the 50 mm/s recommended in the Wiki Page
B) by changing this value you can solved the issue temporarly, without giving the clue to the user

mhaas, 2022-11-02:
Thanks for the explanation! Makes sense :)

closed 2022-11-02 by mhaas