5axes/Calibration-Shapes #130: RetractTower Script is doing nothing

seemingly-ai, 2023-01-11:
Hello,
I just finisched printing a retraction test and was wondering why I had havier stringing in my part then befor and that on every level of the tower. I then opened the .gcode file for the print and looked for all retraction moves by using regex search function "G1 F.... E....". I looked through all apperences and found that all had the same value for F.

My settings in the script are as follows:
[image]

Here are some excerpts from the gcode. (I dont know how to attach the file, if some one can tell me how I can add it to the issue):
This is the layer shift between layer 14 and 15, my retraction distance is 8mm. I started with 20 mm/s (Cura Settings) and thats exactly the value to see in this layer... sadly in every other layer too.

`G1 X55.771 Y44.667 E201.44233
G1 X55.609 Y44.166 E201.45984
G1 X55.581 Y43.883 E201.4693 ; <- last extrusion on layer 14
G0 F1243.7 X55.554 Y43.607
G0 X55.609 Y43.086
G0 F7200 X55.788 Y42.546
G0 X55.905 Y42.346
G1 F1200 E193.4693 ; <- retraction
G1 F300 Z3.2
;MESH:NONMESH
G0 F7200 X30.821 Y40.294 Z3.2
;TIME_ELAPSED:430.404810
;LAYER:15
;TYPE:WALL-INNER
;MESH:RetractTower
G1 F300 Z3.2
G1 F1200 E201.4693 ; <- extrusion after retraction
G1 F1029.3 X30.821 Y40.563 E201.48131
G1 F1181.1 X30.753 Y41.226 E201.50725`

Layer 72 
`G1 X55.609 Y43.086 E560.83095
G1 X55.788 Y42.546 E560.84988
G1 X56.05 Y42.097 E560.86717
G1 X56.243 Y41.885 E560.8767 ; <- last extrusion befor retraction
G0 F1181.3 X56.427 Y41.682
G0 X56.853 Y41.372
G0 F7200 X57.348 Y41.152
G0 X57.601 Y41.099
G1 F1200 E552.8767  ; <- retraction
G0 F7200 X57.54 Y40.805
G0 X56.732 Y41.165
G0 X56.252 Y41.514
G0 X54.655 Y43.626
G1 F1200 E560.8767 ; <- extrusion after retraction
G0 F7200 X54.355 Y43.628
G1 F1312.6 X54.438 Y44.418 E560.90312
G1 X54.683 Y45.175 E560.92959
G1 X55.082 Y45.861 E560.95598
G1 X55.609 Y46.446 E560.98217`

Layer shift from 174 to 175:
`G1 X56.049 Y45.153 E1207.77014
G1 X55.771 Y44.667 E1207.78876
G1 X55.609 Y44.166 E1207.80627
G1 X55.581 Y43.883 E1207.81573 ; <- last extrusion on layer 174
G0 F1235.4 X55.554 Y43.607
G0 X55.609 Y43.086
G0 F7200 X55.788 Y42.546
G0 X55.905 Y42.346
G1 F1200 E1199.81573  ; <- retraction
G1 F300 Z35.2
;MESH:NONMESH
G0 F7200 X31.28 Y40.753 Z35.2
;TIME_ELAPSED:2951.286876
;LAYER:175
;TYPE:WALL-INNER
;MESH:RetractTower
G1 F300 Z35.2
G1 F1200 E1207.81573 ; <- extrusion after retraction
G1 F1165.7 X31.649 Y40.631 E1207.83095
G1 F1365.8 X34.173 Y40.626 E1207.91532`

Like I said if I know how I can upload the file and if it´s needed I can provide more information about machine, printer and software used.

5axes, 2023-01-11:
looks like an issue with your GCode Documentation Script. Try to use only the retract Tower and the Thumbail. And Move the Retract tower script in the first position.

seemingly-ai, 2023-01-11:
Ok that fixed it

9jS2PL5T, 2023-03-13:
I do  have the same problem, but used only the Retract Tower.

9jS2PL5T, 2023-03-13:
Sure and thank you for helping. I'm not familiar with all those code in the file, so it is just blowing my mind. I don't know if the problem is in the code, but i noticed that my tower is looking the same all the way.
[Preset Retract Tower - Retract Distance 1-6_2.txt](https://github.com/5axes/Calibration-Shapes/files/10961010/Preset.Retract.Tower.-.Retract.Distance.1-6_2.txt)

5axes, 2023-03-13:
Your GCode have been Generated with the Plugin : https://github.com/kartchnb/AutoTowersGenerator.  You are not at the right place and  the code is correct. 
[image]

5axes, 2023-03-23:
I think I can Close It

closed 2023-03-23 by 5axes