5axes/Calibration-Shapes #133: MaxFlow missing

onolox, 2023-03-23:
The max flow test menu, stl and .py is missing.

5axes, 2023-03-23:
? Whare are they missing , I dont't understand your issue ?

onolox, 2023-03-23:
Updated question.

5axes, 2023-03-23:
Not really more explicit. 

- Version of the plugin  ?
- Installed from where
- Which version of Cura/ Os.
- A screenshot to show the menu ...  

Well some usefull informations to analyse your "issue"

onolox, 2023-03-23:
Latest of all.
2.2.0
Cura 5.2 or 5.3
installed from marketplace of cura
windows 10

I have corrected it myself here, so the item in the menu appears here: "Maxflow test":

[image]