5axes/Calibration-Shapes #143: Help interpreting maxflow results

Sandrock431, 2023-09-06:
Hello. I'm having trouble interpreting my maxflow results. The example from the wiki makes sense as it seems to have uniform degradation of the flow at a certain layer; however, my test print is fine all the way to the top in most regions, but mainly in the turns it's degraded from top to bottom.

[image]

5axes, 2023-09-06:
printing parameters ?

Sandrock431, 2023-09-07:
My apologies. This was with the base settings suggested on the wiki page, so:
 - Nozzle Size          : 0.4
 - Line Width           : 0.4
 - Layer Height         : 0.25
 - Minimum Layer Time   : 10 s
 - Build Plate Adhesion : Brim
 - Print Speed          : 100 mm/s

5axes, 2023-09-07:
Could be an issue with your initial Flow calibration. Have you made some test to validate the E step calibration on your printer. And have you calibrate your Flow ?

5axes, 2023-09-08:
Looking at the picture, we can also see that the lack of material seems to be present at every acceleration. Perhaps you have a problem with an acceleration too high for your machine? Try lowering the acceleration values for a test run: 
[image]

Sandrock431, 2023-09-12:
I'm going back through calibration. I'll update here one way or another (and thanks for the help!).

Sandrock431, 2023-09-12:
Ugh. I've gone through and re-calibrated everything including `rotation_distance` (22.919) and flow (79.36%), and it's still coming out messy like in the original picture.  I even turned acceleration down to 500mm/s

5axes, 2023-09-13:
Never heard about rotation_distance ??? And concerning the Flow it's not normal to use a value not close to 100% , most of the time this value is  used to fine tune the flow so it's around 97%-102%. If you use a value under 80% it's certainly a problem of method to calculate the Flow ( using a cube with just one Wall and considered that this thickness must be equal to the line thickness, if it s your case, it s not the right method with Cura). You have perhaps a problem of under-extrusion in this case.

Sandrock431, 2023-09-13:
I should have included I was using Klipper from the beginning haha. Thanks for the input. I have re-adjusted my slicer flow (105%ish) which has helped a little bit, but it's still being dumb. I'll go ask in the klipper discord for further advice. Thanks again!

closed 2023-09-13 by Sandrock431