5axes/Calibration-Shapes #147: RetractTower g-code post-processing script configured for Distance results in large extrusion at end of print

dpmott, 2024-02-16:
RetractTower g-code post-processing script configured for Distance results in large extrusion at end of print

Steps to reproduce:
- Cura -> Extensions -> Part for Calibration -> Add a Retract Tower
- Cura -> Extensions -> Post Processing -> Modify G-Code
- Add a script -> RetractTower
- Configure as follows:
  - Command: Distance
  - Starting value: 2
  - Value Increment: 2
  - Change Layer: 38
  - Change Layer Offset: 4
  - Display details on LCD: check
- Slice model and save gcode
- Inspect gcode for the following two lines very near the end of the script:
```
G92 E10;(Set extruder to 10)
G1 F200 E1335.94980 ; Modi
```

These two lines, taken together, will result in the printer attempting to turn the extruder motor roughly 1325 steps at the end of the print.

I believe that this happens because my Cura Machine Settings End G-Code is configured as follows:
```
G0 X0 Y120;(Stick out the part)
M190 S0;(Turn off heat bed, don't wait.)
G92 E10;(Set extruder to 10)
G1 E7 F200;(retract 3mm)
M104 S0;(Turn off nozzle, don't wait)
M84;(Turn off stepper motors.)
```

Note that the line "G1 E7 F200" has been rewritten.

When the RetractTower script is configured with Command: Speed, it rewrites the same line, but with far less disasterous results:
```
G92 E10;(Set extruder to 10)
G1 F4800 E7.00000
```

5axes, 2024-02-16:
Why do you use a these "complexe" notation ?
Even if it's not wrong and you are free to use your own value . It will be the same to write :
G92 E0
G1 E-3 F200

And in this case it's much more "easy" to ready the Gcode and the -3mm retract. And most of all in this case the script should be Ok.

dpmott, 2024-02-21:
> Why do you use this "complex" notation ? 

I created a new printer profile in Cura, and Cura provided that End g-code.  It generally works so I have had no reason to revisit it.

Now that I'm looking through my various printer profiles in Cura, they all do it differently.  It would seem that each printer profile has opinions on how to do this.  The code above is generated as part of creating a "Malyan/Malyan M200 printer" or "Monoprice/Monoprice Select Mini V1" profile.